dji-sdk/Tello-Python #4: Can't get Single_Tello_Test working with a Tello EDU

clydemcqueen, 2018-12-10:
Hi, I'm unable to get the Single_Tello_Test working with my Tello EDU drone. This works fine with my older original Tello.

Here's the transcript:

`clyde@fastr:~/projects/Tello-Python/Single_Tello_Test (master)$ python2 tello_test.py command.txt 
sending command: command to 192.168.10.1
from ('192.168.10.1', 8889): ok
Done!!! sent command: command to 192.168.10.1
sending command: takeoff to 192.168.10.1
from ('192.168.10.1', 8889): error
Done!!! sent command: takeoff to 192.168.10.1
delay 5.0
sending command: land to 192.168.10.1
from ('192.168.10.1', 8889): error
Done!!! sent command: land to 192.168.10.1

id: 0
command: command
response: ok
start time: 2018-12-10 15:04:13.334845
end_time: 2018-12-10 15:04:13.382918
duration: 0.048073

id: 1
command: takeoff
response: error
start time: 2018-12-10 15:04:13.383024
end_time: 2018-12-10 15:04:13.433856
duration: 0.050832

id: 2
command: land
response: error
start time: 2018-12-10 15:04:18.439188
end_time: 2018-12-10 15:04:18.482665
duration: 0.043477
`

clydemcqueen, 2018-12-10:
Ah, never mind. I believe that the drone wasn't "activated." I finally got it to work with the Android app, and now it works with the SDK as well. Closing.

closed 2018-12-10 by clydemcqueen

Gaurav-Gosain, 2021-09-13:
> Ah, never mind. I believe that the drone wasn't "activated." I finally got it to work with the Android app, and now it works with the SDK as well. Closing.

How did you get it "activated", I am struggling with mine right now

TheGreatRambler, 2022-03-10:
I would also like to know how you activate it, the mobile app crashes on startup so I can't activate from there