dji-sdk/Tello-Python #6: Where is the video protocol documented?

wtesler, 2018-12-31:
https://github.com/dji-sdk/Tello-Python/blob/5b0b8193c4b701c612b73b8c1e59a11046d4506a/Tello_Video/tello.py#L104

Where is this behavior documented (where a packet length not of 1460 is used to partition a frame)? When I do this, I get unexpected output which looks like: 

[image]

If I append conditions to that statement so it looks like 
`if (packet.length != 1460 && packet[0] != 0 && packet[1] != 0 && packet[2] != 0 && packet[3] != 1)`

Then I get an actual picture, but it is choppy and full of artifacts. It looks like:

[image]

Where is the protocol documented and what is wrong with the given implementation?

hanker-lu, 2019-01-11:
> [Tello-Python/Tello_Video/tello.py](https://github.com/dji-sdk/Tello-Python/blob/5b0b8193c4b701c612b73b8c1e59a11046d4506a/Tello_Video/tello.py#L104)
> 
> Line 104 in [5b0b819](/dji-sdk/Tello-Python/commit/5b0b8193c4b701c612b73b8c1e59a11046d4506a)
> 
>  if len(res_string) != 1460: 
> Where is this behavior documented (where a packet length not of 1460 is used to partition a frame)? I'm having trouble finding any official sources which talk about the protocol.

Hello，you can refer to doc/readme.pdf.

johnwalicki, 2019-01-14:
@hanker-lu  - The "What's possible?" thread in the TelloPliots.com forum talks about reverse engineering the Tello protocols (including the video stream). https://tellopilots.com/threads/tello-whats-possible.88/

hanker-lu, 2019-01-25:
> @hanker-lu - The "What's possible?" thread in the TelloPliots.com forum talks about reverse engineering the Tello protocols (including the video stream). https://tellopilots.com/threads/tello-whats-possible.88/
Have already seen it, thank you.

closed 2019-02-11 by hanker-lu