dji-sdk/Tello-Python #8: Tello_video is not working.

zsole19, 2019-01-08:
Is not able to include the package h264 even if is it in the correct folder, do you have any suggestion?

`File "D:\Tello\Tello-Python-master\Tello-Python-master\Tello_Video\tello.py", line 5, in <module>
import libh264decoder
ImportError: DLL load failed: The specified module could not be found.`

hanker-lu, 2019-01-10:
Hello, have you run a one-click installation script？ If so, you can try manually putting h264decoder.pyd in the site-package folder.

kesaroid, 2019-02-04:
> Hello, have you run a one-click installation script？ If so, you can try manually putting h264decoder.pyd in the site-package folder.

@hanker-lu Did you mean libh264decoder.pyd in site-packages?

I am having the same problem importing lib264decoder when I run main.py, Even the install.bat for windows doesn't seem to be working very accurately. So I manually pip installed all the dependencies, but the h264decoder isn't built. 
@hknakata how did you solve the problem?

johnadams2076, 2019-02-04:
Works for me. Check Pose Recognition in video. 
https://www.linkedin.com/feed/update/urn:li:activity:6498225445987852288
Okay.
As a workaround,
For Win-64, following files need to be downloaded into your
 C:\<Path to Tello project root>\Tello-Python\Tello_Video\install\Windows.

> https://www.python.org/ftp/python/2.7.15/python-2.7.15.amd64.msi
> https://bootstrap.pypa.io/get-pip.py
> https://ffmpeg.zeranoe.com/builds/win64/shared/ffmpeg-20180825-844ff49-win64-shared.zip
> https://nchc.dl.sourceforge.net/project/boost/boost-binaries/1.68.0/boost_1_68_0-msvc-12.0-64.exe
> https://download.microsoft.com/download/2/E/6/2E61CFA4-993B-4DD4-91DA-3737CD5CD6E3/vcredist_x64.exe

Type "command" in the search bar and right click. Select "Run as Administrator".
cd to C:\<Path to Tello project root>\Tello-Python\Tello_Video\install\Windows.
type "install"

It will extract and put files in the appropriate directories.
Now check C:\Python27\Lib\site-packages.
Pyd file along with several dlls must be present.
Run main.py.

Hope this helps.

As for 3.6, I tried cmake for h264decoder. I got it compiled. Linker Ld.exe was looking for 3 dlls. I did not go further.
I switched to the above method. 

Lets explore what's possible and beyond.

kesaroid, 2019-02-05:
Still no luck. The install.bat file keeps crashing. Opens command prompt, asks for the admin rights and then closes every time. I have tried everything. Maybe it's a bug in the installation file itself? Did you get it to work with win-64 itself?

johnadams2076, 2019-02-06:
Unpack ffmpeg-20180825-844ff49-win64-shared.zip
Unpack boost_1_68_0-msvc-12.0-64.exe

Go to C:\\\<Path to extracted ffmpeg\bin

Copy all dlls to C:\Python27\Lib\site-packages

Go to C:\<Path to extracted Boost >\boost_1_68_0\lib64-msvc-12.0

Copy boost_python27-vc120-mt-x64-1_68 to C:\Python27\Lib\site-packages

Copy C:\<Path to project>\Tello-Python\Tello_Video\h264decoder\Windows\x64\libh264decoder.pyd to C:\Python27\Lib\site-packages

Now run main.py.
Let me know.

kesaroid, 2019-02-06:
Did the exact same thing, but unfortunately ..
`Traceback (most recent call last):
  File "main.py", line 1, in <module>
    import tello
  File "C:\Users\Kesar\Documents\Artificial intelligence\Dji Tello\Tello-Python\Tello_Video\tello.py", line 5, in <module>
    import libh264decoder
ImportError: DLL load failed: The specified module could not be found.`
the same error.
The libh264decoder.pyd file doesn't seem to work right for me

johnadams2076, 2019-02-06:
Is your python Base interpreter pointing to C:\Python27\python.exe ?

kesaroid, 2019-02-06:
> Is your python Base interpreter pointing to C:\Python27\python.exe ?

yes, yes it does. what else could I try to figure this out?

hanker-lu, 2019-02-13:
> > Is your python Base interpreter pointing to C:\Python27\python.exe ?
> 
> yes, yes it does. what else could I try to figure this out?
> Should I try installing python for x86 and try h264decoder with that?

For win64,i have pushed new commit about some dll needed to run tello_video.You can try to copy all these .dll/.pyd to your site-package file and run the program again.

hanker-lu, 2019-02-13:
> For win64,i have pushed new commit about some dll needed to run tello_video.You can try to copy all these .dll/.pyd to your site-package file and run the program again.

In addition ，you should run the vcredist_x64.exe to setup the vs2013 dependency.

kesaroid, 2019-02-13:
@hanker-lu Thank you so much. It's working just fine except for a minor h264decoder error
`[h264 @ 0000000013786800] non-existing PPS 0 referenced
[h264 @ 0000000013786800] decode_slice_header error
[h264 @ 0000000013786800] no frame!`

zsole19, 2019-02-16:
Hello, after copying all these dll, I still get the same error. 
`import libh264decoder
ImportError: DLL load failed: The specified module could not be found.`
I am running windows x64.

hanker-lu, 2019-02-18:
> Hello, after copying all these dll, I still get the same error.
> `import libh264decoder ImportError: DLL load failed: The specified module could not be found.`
> I am running windows x64.

Hi， please check if you run the vcredist_x64.exe to setup the vs2013 dependency.

zsole19, 2019-02-18:
> Hi， please check if you run the vcredist_x64.exe to setup the vs2013 dependency.

@hanker-lu, yes I have done that. I have done everything that is written here above. I don't know what else could I do.

zsole19, 2019-02-18:
Should I try to build the h264 library for my self?

sriranjanr, 2019-04-16:
Getting the following error. I copied libh264decoder to site-packages in python folder but still the import module error 

`in <module>
    import libh264decoder
ImportError: DLL load failed: The specified module could not be found.`

sriranjanr, 2019-04-16:
I ran the install.bat file. It was unable to install pip, I ran get-pip.py and then it installed pip. Also it was unable to install boost. So I downloaded boost and boost is just like a large collection of files which got stored in c:/local/boost1_68_0. What should I do with these boost files? I am still getting the same import libh264decoder error. Is there not an easier way to read images one by one from the Tello edu drone..?

sriranjanr, 2019-04-19:
Could anyone help me to get the tello_video program running. I get libh264 error?

Qlinkwisp, 2019-05-12:
> Works for me. Check Pose Recognition in video.
> https://www.linkedin.com/feed/update/urn:li:activity:6498225445987852288
> Okay.
> As a workaround,
> For Win-64, following files need to be downloaded into [...]

no more  ffmpeg-20180825-844ff49-win64-shared.zip on ffmpeg.zeranoe.com

https://ffmpeg.zeranoe.com/builds/win64/shared/ffmpeg-20180825-844ff49-win64-shared.zip

i think install.bat should update

gearsmotion789, 2019-05-18:
I am getting "libh264decoder" error as well. I also agree that the ffmpeg is not downloaded correctly, and that install.bat needs to be updated.

[image]

AlienKevin, 2019-06-09:
I also encountered the same issue with ffmpeg. In addition, I found that libboost is not installing properly as well.

[image]

Erro message "msxml3.dll: Access is denied" persists even after I removed md5 testing.

psychon7, 2019-06-26:
How to recive tello video stream in windows? MD5 mismatch, decoder error and this is still not perfect. Please someone from tello community make it accessible

joeyc002, 2019-07-09:
I bought the Tello so I had an excuse to play around with Python, but this has completely put me off. Why is Python so difficult to get working? It's way over complicated.

joeyc002, 2019-07-09:
Ok, I fixed this by extracting tello_video_dll(ForWin64).zip which is in the root directory of this project into C:\Python27\Libs\site-packages. 

Make sure you're using x64 of all your installations if you still have issues. I had x86 of python which caused problems for me.

projectdrone2, 2019-10-13:
@kesaroid , I have the same issue "[h264 @ xxx] non-existing PPS 0 referenced [...]". Have you managed to find a fix?

cozmobotics, 2019-12-29:
I have the same problem. 
After trying in Windows,I tried in a Virtualbox where I have installed Mint 19_1. 
I started the Linux_install.sh. 
I get a lot of messages, in the end I got: 

```
-- Configuring incomplete, errors occurred!
See also "/media/sf_D_DRIVE/daten/robotics/tello/dji-sdk/Tello-Python-master/Tello_Video/h264decoder/build/CMakeFiles/CMakeOutput.log".
See also "/media/sf_D_DRIVE/daten/robotics/tello/dji-sdk/Tello-Python-master/Tello_Video/h264decoder/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
cp: cannot stat 'libh264decoder.so': No such file or directory
Compilation and Installation Done!
```
I attach both logfiles. 
CMakeError.log begins with 

```
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: CMAKE_CXX_COMPILER-NOTFOUND 
Build flags: 
Id flags:  

The output was:
No such file or directory
```

 This block repeats again and again, with different Id flags. 

[CMakeError.log](https://github.com/dji-sdk/Tello-Python/files/4007931/CMakeError.log)
[CMakeOutput.log](https://github.com/dji-sdk/Tello-Python/files/4007932/CMakeOutput.log)

fidsusj, 2020-05-18:
I'm quoting from the FAQ:

In the latest commit, a zip named 'tello_video_dll.zip' is provided (for win64 only), which includes the required files related to boost, ffmpeg, libh264decoder, vs2013 compiler.If you want to install it:
First, double-click to run vcredist_x64.exe (vs2013 compiler installation package) to install.Second, copy the remaining files in the folder to the C:\Python27\Lib\site-packages path. (With this zip, you can avoid the online download and partial installation, but note that the files in this zip are only available for win64! nAlso, like numpy, matplotlib, pillow and opencv-python you still need to install with pip by yourself.)

This might help some of you guys

closed 2023-09-01 by zsole19

Kentin01, 2024-08-16:
> I'm quoting from the FAQ:
>
> In the latest commit, a zip named 'tello_video_dll.zip' is provided (for win64 only), which includes the required files related to boost, ffmpeg, libh264decoder, vs2013 [...]

Hi
Where can I find 'tello_video_dll.zip ??

Kentin01, 2024-08-16:
> > Works for me. Check Pose Recognition in video.
> > https://www.linkedin.com/feed/update/urn:li:activity:6498225445987852288
> > Okay.
> > As a workaround,
> > For Win-64, following files need to be downloaded into your
> > C:\Tello-Python\Tello_Video\install\Windows.
> > > https://www.python.org/ftp/python/2.7.15/python-2.7.15.amd64.msi
> > > https://bootstrap.pypa.io/get-pip.py
> > > https://ffmpeg.zeranoe.com/builds/win64/shared/ffmpeg-20180825-844ff49-win64-shared.zip
> > > https://nchc.dl.sourceforge.net/project/boost/boost-binaries/1.68.0/boost_1_68_0-msvc-12.0-64.exe
> > > https://download.microsoft.com/download/2/E/6/2E61CFA4-993B-4DD4-91DA-3737CD5CD6E3/vcredist_x64.exe
> > 
> > 
> > Type "command" in the search bar and right click. Select "Run as Administrator".
> > cd to C:\Tello-Python\Tello_Video\install\Windows.
> > type "install"
> > It will extract and put files in the appropriate directories.
> > Now check C:\Python27\Lib\site-packages.
> > Pyd file along with several dlls must be present.
> > Run main.py.
> > Hope this helps.
> > As for 3.6, I tried cmake for h264decoder. I got it compiled. Linker Ld.exe was looking for 3 dlls. I did not go further.
> > I switched to the above method.
> > Lets explore what's possible and beyond.
> 
> no more ffmpeg-20180825-844ff49-win64-shared.zip on ffmpeg.zeranoe.com
> 
> https://ffmpeg.zeranoe.com/builds/win64/shared/ffmpeg-20180825-844ff49-win64-shared.zip
> 
> i think install.bat should update

where can I find [ffmpeg.zeranoe.com](https://ffmpeg.zeranoe.com/builds/win64/shared/ffmpeg-20180825-844ff49-win64-shared.zip)

Kentin01, 2024-08-16:
[image]
Hi community
I'm experiencing this issue how to solve it please