dji-sdk/Tello-Python #12: install.bat file keeps crashing on win-64

kesaroid, 2019-02-05:
as [mentioned](https://github.com/dji-sdk/Tello-Python/issues/8#issuecomment-460533683) earlier, install.bat and install_manual.bat closes the cmd prompt after admin rights. I have manually tried installing all dependencies, but I need h264decoder to be built. 
Any suggestions?

hanker-lu, 2019-02-13:
> as [mentioned](https://github.com/dji-sdk/Tello-Python/issues/8#issuecomment-460533683) earlier, install.bat and install_manual.bat closes the cmd prompt after admin rights. I have manually tried installing all dependencies, but I need h264decoder to be built.
> Any suggestions?
For win64,i have pushed new commit about some dll needed to run tello_video.You can try to copy all these .dll/.pyd to your site-package file and run the program again.In addition ，you should run the vcredist_x64.exe to setup the vs2013 dependency.

kesaroid, 2019-02-13:
Thanks @hanker-lu, It's working just fine now

closed 2019-02-13 by kesaroid