dji-sdk/Tello-Python #14: Tello_Video main.py does not render video feed MacOS High Sierra

JimiPedros, 2019-02-08:
Hi there, I have installed the project successfully using the `./install.sh` script.

files in Tello_Video: 
```
LICENSE.md		__pycache__		h264decoder		install			main.py			tello.pyc		tello_control_ui.pyc
README.md		get-pip.py		img			libh264decoder.so	tello.py		tello_control_ui.py	venv
```

```
python main.py 
sent: command
sent: streamon
```

[image]

[image]

when I run `main.py` the control window opens fine but no video feed appears.

hanker-lu, 2019-02-13:
Hi.Normally, when you run python main.py, the real-time video will be displayed directly on the GUI interface, and you can take photos. So this may be because the video stream is blocked when tello transmits video streams to your PC device. So, you need to check if your PC network settings have disabled the UDP port 11111 for video streaming.

schnellerblitz99, 2019-03-03:
Hey, I have the same issues. I get the UI but nothing seems to work (Mojave, Python 2.7.15) It seems that my tello has not connected correctly since the LED is orange flashing. I turned my firewall off, so that port 11111 is enabled, but it did not work out. Have you guys another idea? thanks a lot!

schnellerblitz99, 2019-03-03:
Hey, I found the solution for my problem: Updating the firmware of the tello drone. After having installed the new firware with the tello app, it worked out fine. Though the control of the drone seems to be a bit rough.