dji-sdk/Tello-Python #15: Error when taking snapshot - `Exception in Tkinter callback`

JimiPedros, 2019-02-08:
Error log:

```
Exception in Tkinter callback
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk/Tkinter.py", line 1536, in __call__
    return self.func(*args)
  File "/Users/jimipedros/git/Tello-Python/Tello_Video/tello_control_ui.py", line 245, in takeSnapshot
    cv2.imwrite(p, cv2.cvtColor(self.frame, cv2.COLOR_RGB2BGR))
error: OpenCV(4.0.0) /Users/travis/build/skvark/opencv-python/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```

hanker-lu, 2019-02-13:
Hi.Normally, when you run python main.py, the real-time video will be displayed directly on the GUI interface, and you can take photos. So this may be because the video stream is blocked when tello transmits video streams to your PC device. So, you need to check if your PC network settings have disabled the UDP port 11111 for video streaming.

cosminv6u, 2019-04-15:
i have the same problem ..

KeitaIto123, 2019-06-22:
I also have the same problem. (on Ubuntu 16.04.5 LTS)
I tried again after executing commands as below to enable the UDP port 11111, but only to remain unchanged.
```
$ sudo ufw enable
$ sudo ufw allow 11111/udp
$ sudo ufw reload
```

SanguineXT, 2019-11-15:
Hi, have you solved it out?