dji-sdk/Tello-Python #17: Moving to python3

butterfly-pn, 2019-02-25:
Python 2.7 will soon be deprecated. Have you considered moving the project to python3?

f41ardu, 2019-03-17:
Does anybody knows how to modify the h264decoder to work with phyton3.x ?

f41ardu, 2019-03-30:
Solved. I've moved Tello_Video from python2.7 to python3.5 (on Raspberry PI). 
I'll prepare a pull  cause I have to do a cleanup in my repository and integrate the h264 library based on the development branch. So stay tuned. At least and after some research it was pretty easy.

mike-shock, 2019-03-31:
I'd love to control my Tello from Raspberry Pi in Python3!

cjs123456, 2019-12-01:
How could I get the libh264decoder.pyd file in the windows environment? I want to run tello in the windows environment, what should I do？I am completely obsessed with it for several time, please help me.@f41ardu

f41ardu, 2019-12-25:
Find out how to build the h264 decoder library on a Windows client. My version will work on Rasperrypi (tested) and Linux clients (not tested yet). I'm sorry to say and looking for myself to get a version running on Windows 10.