dji-sdk/Tello-Python #19: Tello SDK cannot go move after take off

yeohkc1995, 2019-03-16:
Hi, the I am using this sdk for my newly bought tello. The video feed is fine, the take off and landing command works, and the rotate cw and ccw command works "Response: ok". However, when I send the any other commands (forward, backward, left, right etc) the command always return a "Response: Out of Range", and consequently my Tello can only take off, rotate on the spot, and land....

But it shouldn't be a connection issue, as it can recognize the rotate, take off and landing perfectly. 

And the drone works perfectly well on my tello app in all aspects (if this info helps)

A example of the command from the tello.py file that does not work is this

    def move_backward(self, distance):
        """Moves backward for a distance.

        See comments for Tello.move().

        Args:
            distance (int): Distance to move.

        Returns:
            str: Response from Tello, 'OK' or 'FALSE'.

        """

        return self.move('back', distance)

I didnt change the code, so it is as it originally was.

Anyone has any idea whats the issue :(   its been bugging me. Any help will be appreciated!

sgolodetz, 2019-04-02:
I'm exploring the possibility that the distance it's trying to move by default is too small (my Tello's just run out of battery, so more on this when it's recharged :)). If you look at this line in `tello_control_ui.py`, it's trying to move 10cm by default:

    self.distance = 0.1  # default distance for 'move' cmd

However, if you look at the Tello SDK documentation, the smallest distance you can ask it to move is 20cm:

https://dl-cdn.ryzerobotics.com/downloads/tello/20180910/Tello%20SDK%20Documentation%20EN_1.3.pdf

I suspect that if you change it from `0.1` to `0.2`, it will work - but I can't try that till my battery's charged :)

sgolodetz, 2019-04-02:
Yup, that works, for the record. Although the UI did later crash on me, for what I suspect are unrelated reasons.

yeohkc1995, 2019-04-10:
Hi all thanks so much for the help. I have not logged in in while. Allow me a few days to check it out (am busy with work now). really hope it works... I will get back to this asap. Thanks yall so much, appreciate it!! :)