dji-sdk/Tello-Python #20: Tello_Video on Phyton3 is working now.

f41ardu, 2019-03-31:
Feel free to test https://github.com/f41ardu/Tello-Python/tree/master/PY3_Tello_Videop
On an Raspberry PI 3+ I still got multiple video errors but video is working n python 3.5. I assume it may work on 3.x, x >= 5 as well. Please let me know if it is working an I'll request a merge into dji-sdk/Tello-Phyton.

rum-neutralo, 2019-04-07:
@f41ardu I tried your code, but had to use "Tkinter" (with uppercase T) as import statement in your file PY3_Tello_Videop/tello_control_ui. ! I`m on MAC OS.

f41ardu, 2019-05-02:
> 
> 
> @f41ardu I tried your code, but had to use "Tkinter" (with uppercase T) as import statement in your file PY3_Tello_Videop/tello_control_ui. ! I`m on MAC OS.

Seems to be MAC OS specific.

boahneedle, 2019-05-08:
I'm getting:  

`ImportError: dlopen(/Users/noahnoah/Documents/Tello-Python-master 2/PY3_Tello_Videop/libh264decoder.so, 2): no suitable image found.`

upon running your fork with Py 3.7.3 on MacOS Sierra 10.12.2

f41ardu, 2019-05-08:
Build libh264decoder.so and place it in 

pi@raspberrypi:~/code/Tello-Python/PY3_Tello_Videop $ ls
h264decoderpi3     LICENSE.md  __pycache__  tello_control_ui.py
**libh264decoder.so**  main.py     README.md    tello.py

boahneedle, 2019-05-08:
Hello, thank you for your reply.

I placed the libh264decoder.so file that is included in your repo into that _exact_ directory, and that is when the above error was thrown.

For clarification:  do I have to compile from source?

f41ardu, 2019-05-08:
Sorry, yes you have to build the decoder lib. Unfortunately I included the Raspberry binaries.

kekoasantana, 2019-07-09:
Hello, I have been getting the issue 

File "main.py", line 1, in <module>
    import tello as tello
  File "/Users/KekoaSantana/Documents/tellopython3/PY3_Tello_Videop/tello.py", line 5, in <module>
    import libh264decoder
ImportError: dynamic module does not define module export function (PyInit_libh264decoder)

I am not sure what the cause of this is and looked it up and couldn't find much. maybe you are aware of it? I did the whole install thing and then moved the libh264decoder.so from tello_video to py3_tello_videop since there is no install in that folder.