dji-sdk/Tello-Python #24: RuntimeError: cannot open context

martsalz, 2019-05-17:
`python main.py 
[indeo3 @ 0x1360440] Invalid picture dimensions: 0 x 0!
Traceback (most recent call last):
  File "main.py", line 14, in <module>
    main()
  File "main.py", line 7, in main
    drone = tello.Tello('', 8889)  
  File "/home/martin/Schreibtisch/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello.py", line 25, in __init__
    self.decoder = libh264decoder.H264Decoder()
RuntimeError: cannot open context
Exception AttributeError: "Tello instance has no attribute 'socket'" in <bound method Tello.__del__ of <tello.Tello instance at 0x7f02518d68c0>> ignored
`
Any idea?

Thanks

guidoreboredo, 2019-10-17:
Could you solve this?

martsalz, 2019-10-18:
Sorry, I have sold the drone and I am not involved in the project anymore...