dji-sdk/Tello-Python #26: Feature Request: Video Stream with Separate AP

alecGraves, 2019-05-18:
The limitation is mentioned in [the FAQ](https://github.com/dji-sdk/Tello-Python/blob/693776d65b691525773adf4f320f034946fb15b2/TelloPython_FAQ.txt): 
```
Q8:Can Tello EDU transmit video streams externally in station mode (send ap 'ssid' 'password' command for Tello EDU)?

A8:This feature is currently not supported.
```

Can this feature could be added to the SDK? It would make the EDU edition much more useful for may lab. 

Currently, this limitation removes all benefits of having the EDU edition because I need live video from all four drones.

TobiMiller, 2019-06-07:
I agree. I thought this would be possible. That is the reason why we bought three of them.

Brownie79, 2020-01-16:
Came here to make this issue +1

alecGraves, 2020-01-17:
As a kind of work around, you can use multiple usb wi-fi adapters and some linux networking to recieve multiple video streams from a single computer.

See my fork for implementation: https://github.com/alecGraves/Tello-Python