dji-sdk/Tello-Python #31: TypeError: __init__() should return None, not 'NoneType'

saiivarma, 2019-09-01:
I'm Getting the following error while running main.py 

Traceback (most recent call last):
  File "main.py", line 14, in <module>
    main()
  File "main.py", line 7, in main
    drone = tello.Tello('', 8889)  
  File "/Users/vsvsvarma/Desktop/Tello-Python/Tello_Video/tello.py", line 25, in __init__
    self.decoder = libh264decoder.H264Decoder()
TypeError: __init__() should return None, not 'NoneType'
Exception AttributeError: "Tello instance has no attribute 'socket'" in <bound method Tello.__del__ of <tello.Tello instance at 0x1237d0d88>> ignored

closed 2019-09-01 by saiivarma

yuxiangdai, 2019-10-07:
Hi, were you able to resolve this issue?

guidoreboredo, 2019-10-18:
Hi do you solved this issue?

saiivarma, 2019-10-22:
Had an issue with the libboost-python and cmake installation. If you’re having multiple versions of python installed you might have to link the correct python version to cmake by changing PYTHON_EXECUTABLE,PYTHON_INCLUDE_DIR and PYTHON_LIBRARY Variables in CmakeCache.txt

pringithub, 2020-03-19:
How is this done? I can't find a CmakeCache.txt anywhere ..

interactivetech, 2020-05-21:
@pringithub the CMakeCache.txt is in the h264decoder/build/ after you run the command `cmake ..`. @saiivarma worked for me by editing the PYTHON_EXECUTABLE,PYTHON_INCLUDE_DIR and PYTHON_LIBRARY Variables then doing `make`

csmaster23, 2020-07-19:
@interactivetech Sorry but what do I change them to? Currently, PYTHON_INCLUDE_DIR and PYTHON_LIBRARY are going to some Xcode paths and PYTHON_EXECUTABLE is through /usr/local/bin/python2.7....just wondering what I switch the paths to.