dji-sdk/Tello-Python #37: Setting up in installation

hkpc0715, 2019-09-26:
I have followed the following steps:

Mac

Make sure you have the latest Xcode command line tools installed. If not, you might need to update your OS X and XCode to the latest version in order to compile the h264 decoder module
Go to the "install\Mac" folder folder in command line, run
   chmod a+x ./mac_install.sh
   ./mac_install.sh

However, my installation said:

Configuring incomplete, errors occurred!
See also ''User/hkpc/downloads/Tello-Python-master/Tello_Video/h264decoder/build/CMakeFiles/CMakeOutput.log''.
make: *** No targets specified and no makefile found. Stop.
cp: libh264decoder.so: No such file or directory
Compilation and Installation Done!

Can I know what am I missing and how to fix this?

dji-dev, 2019-09-26:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/28330'>Zendesk ticket #28330</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

https://github.com/dji-sdk/Tello-Python/tree/master/Tello_Video/install/Mac
You can download the files in this directory and automatically install the test environment.

Best Regards,

Ho Yan
RYZE TECHNICAL SUPPORT

hkpc0715, 2019-09-26:
> Public comment from Ho.Yan in [Zendesk ticket #28330](https://djisdksupport.zendesk.com/agent/tickets/28330):
> 
> Dear Developer,
> 
> Thanks for reaching Ryze!
> 
> https://github.com/dji-sdk/Tello-Python/tree/master/Tello_Video/install/Mac
> You can download the files in this directory and automatically install the test environment.
> 
> Best Regards,
> 
> Ho Yan
> RYZE TECHNICAL SUPPORT

Hi Ho Yan,

Thank you for your reply. However, I think you have misunderstand my meaning. I have tried to download the above files and run in my terminal. Where I have got the following errors:

Configuring incomplete, errors occurred!
See also ''User/hkpc/downloads/Tello-Python-master/Tello_Video/h264decoder/build/CMakeFiles/CMakeOutput.log''.
make: *** No targets specified and no makefile found. Stop.
cp: libh264decoder.so: No such file or directory
Compilation and Installation Done!

georgegoglodze, 2019-11-17:
Same error here on OS X and XCode

manujimele, 2020-05-29:
> Hi Ho Yan,
>
> Thank you for your reply. However, I think you have misunderstand my meaning. I have tried to download the above files and run in my terminal. Where I have got the following errors: [...]

@hkpc0715 same issue here... did you solve it?