dji-sdk/Tello-Python #39: Pose recognition doesn't work

pet93, 2019-10-02:
Hi!
I'm attempting to use pose recognition, but when i run the main project and i switch on the pose recognition, the joints' skeleton of my body doesn't appear. I've downloaded the model and followed instructions written in the readme. Any solution?

dji-dev, 2019-10-08:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/28536'>Zendesk ticket #28536</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

Please check to see if "Pose Recognition Status: On" is turned on after running "main.py".

Best Regards,
Ho Yan
RYZE TECHNICAL SUPPORT

kuengla, 2019-10-08:
I get that error when trying to switch on the pose recognition. My OS is Linux.

```
[ INFO:0] Initialize OpenCL runtime...
OpenCV Error: Assertion failed (output_slice.isContinuous() && output_slice.size == curr_output.size) in fuseLayers, file /io/opencv/modules/dnn/src/dnn.cpp, line 1430
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/lars/Documents/software_dev/tello/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello_control_ui.py", line 113, in videoLoop
    cmd,self.draw_skeleton_flag,self.points = self.my_tello_pose.detect(self.frame)
  File "/home/lars/Documents/software_dev/tello/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello_pose.py", line 209, in detect
    output = self.net.forward()
error: /io/opencv/modules/dnn/src/dnn.cpp:1430: error: (-215) output_slice.isContinuous() && output_slice.size == curr_output.size in function fuseLayers
```

pet93, 2019-10-10:
Hi, I've discovered that actually it works but it is very very slow in processing frames. Any suggest for improving speed?
Thank you

dji-dev, 2019-10-11:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/28536'>Zendesk ticket #28536</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

The sample we provide is for reference only. If you need to increase the processing efficiency, you need to modify it yourself.

Best Regards,
Ho Yan
RYZE TECHNICAL SUPPORT

145594, 2020-02-05:
There is no pose_iter_160000.caffemodel file under the path.
Tello-Python-master\Tello_Video_With_Pose_Recognition\model\pose\mpi\pose_iter_160000.caffemodel.
I checked the package you provided carefully, but can't find.

frank804, 2020-10-08:
@kuengla  do you solve this problem? help!