dji-sdk/Tello-Python #40: Tello_Video & Tello_Video_With_Pose_Recognition

hkpc0715, 2019-10-09:
Hi all,

I was trying out both Tello_Video & Tello_Video_With_Pose_Recognition packages in my **Linux - ubuntu 16.04 LTS**

At first, I installed Tello_Video with the one click install and it has successfully run, however, when i tries to install and run Tello_VIdeo_With_Pose_Recognition, it shows the following errors, and now it no longer runs again.

I have checked and I moved the 'libh264decoder.so' file into my python's site-packages file & /usr/lib/  , the above move was suggested by the internet. But still getting the error:

`spark@spark:~/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition$ python main.py
Traceback (most recent call last):
  File "main.py", line 1, in <module>
    import tello
  File "/home/spark/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello.py", line 5, in <module>
    import libh264decoder
ImportError: libavcodec.so.57: cannot open shared object file: No such file or directory
spark@spark:~/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition$ cd ..
spark@spark:~/Downloads/Tello-Python-master$ cd Tello_Video
spark@spark:~/Downloads/Tello-Python-master/Tello_Video$ python main.py
Traceback (most recent call last):
  File "main.py", line 1, in <module>
    import tello
  File "/home/spark/Downloads/Tello-Python-master/Tello_Video/tello.py", line 5, in <module>
    import libh264decoder
ImportError: libavcodec.so.57: cannot open shared object file: No such file or directory
spark@spark:~/Downloads/Tello-Python-master/Tello_Video$ `

hkpc0715, 2019-10-11:
> Public comment from Ho.Yan in [Zendesk ticket #28662](https://djisdksupport.zendesk.com/agent/tickets/28662):
> 
> Dear Developer,
> 
> Thanks for reaching Ryze!
> 
> Please check if there is a pose_iter_160000.caffemodel file under the path.
> Tello-Python-master\Tello_Video_With_Pose_Recognition\model\pose\mpi\pose_iter_160000.caffemodel
> 
> You can recheck that the development environment is completely installed.
> 
> Best Regards,
> Ho Yan
> RYZE TECHNICAL SUPPORT

Thank you for your reply. I have checked the folder that you mentioned, and I can only find 'pose_deploy_linevec_faster_4_stages.prototest' inside:

Tello-Python-master\Tello_Video_With_Pose_Recognition\model\pose\mpi\

**No pose_iter_160000.caffemodel**

I have installed the Linux version through linux_install.sh, and extracted the following error lines:

root@spark:~/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/install/Linux# chmod +x linux_install.sh
root@spark:~/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/install/Linux# ./linux_install.sh
Compiling and Installing the Tello Video Stream module
You might need to enter your password
Hit:1 http://ppa.launchpad.net/nemh/systemback/ubuntu xenial InRelease         
Hit:2 http://cn.archive.ubuntu.com/ubuntu xenial InRelease                     
Get:3 http://security.ubuntu.com/ubuntu xenial-security InRelease [109 kB]
Get:4 http://packages.ros.org/ros/ubuntu xenial InRelease [4,678 B]            
Err:4 http://packages.ros.org/ros/ubuntu xenial InRelease                      
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Get:5 http://cn.archive.ubuntu.com/ubuntu xenial-updates InRelease [109 kB]    
Get:6 http://cn.archive.ubuntu.com/ubuntu xenial-backports InRelease [107 kB]  
Fetched 330 kB in 3s (94.2 kB/s)                       
Reading package lists... Done
**W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros/ubuntu xenial InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654**
**W: Failed to fetch http://packages.ros.org/ros/ubuntu/dists/xenial/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Some index files failed to download. They have been ignored, or old ones used instead.**
Reading package lists... Done
    
Get:5 http://cn.archive.ubuntu.com/ubuntu xenial-updates InRelease [109 kB]    
Err:3 http://packages.ros.org/ros/ubuntu xenial InRelease                      
  **The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654**

Get:6 http://cn.archive.ubuntu.com/ubuntu xenial-backports InRelease [107 kB]  
Fetched 330 kB in 3s (99.2 kB/s)                       
Reading package lists... Done
**W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros/ubuntu xenial InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654**
**W: Failed to fetch http://packages.ros.org/ros/ubuntu/dists/xenial/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Some index files failed to download. They have been ignored, or old ones used instead.
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/__init__.py:83:** 

-- Configuring done
-- Generating done
-- Build files have been written to: /home/spark/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/h264decoder/build
Scanning dependencies of target h264decoder
[ 33%] Building CXX object CMakeFiles/h264decoder.dir/h264decoder.cpp.o
[ 66%] Building CXX object CMakeFiles/h264decoder.dir/h264decoder_python.cpp.o
[100%] Linking CXX shared library libh264decoder.so
[100%] Built target h264decoder
Compilation and Installation Done!

Furthermore, I have tried to run python main.py, where it shows:

root@spark:~/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition# python main.py
sent: command
sent: streamon
OpenCV Error: Unspecified error (FAILED: fs.is_open(). Can't open "model/pose/mpi/pose_iter_160000.caffemodel") in ReadProtoFromBinaryFile, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/dnn/src/caffe/caffe_io.cpp, line 1126
Traceback (most recent call last):
  File "main.py", line 14, in <module>
    main()
  File "main.py", line 8, in main
    vplayer = TelloUI(drone,"./img/")
  File "/home/spark/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello_control_ui.py", line 43, in __init__
    self.my_tello_pose = Tello_Pose()
  File "/home/spark/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello_pose.py", line 29, in __init__
    self.net = cv2.dnn.readNetFromCaffe(self.protoFile, self.weightsFile)
cv2.error: /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/dnn/src/caffe/caffe_io.cpp:1126: error: (-2) FAILED: fs.is_open(). Can't open "model/pose/mpi/pose_iter_160000.caffemodel" in function ReadProtoFromBinaryFile

hkpc0715, 2019-10-11:
> Public comment from Ho.Yan in [Zendesk ticket #28662](https://djisdksupport.zendesk.com/agent/tickets/28662):
>
> Dear Developer,
>
> Thanks for reaching Ryze!
>
> Please check if there is a [...]

May I know what should I do if the install is failed mentioned above? Shall I download pose_iter_160000.caffemodel file from online and pase it into the path?

hkpc0715, 2019-10-11:
I am now running the getModels.sh in the model file, and it is downloading the file. It is talking about 6 hours with a cable connection (1000mpbs). Anyway, I will try after the download is finished, and will update with you once again. Might still need your help later.

Thank you for your kind help, Dji technical support team!

hkpc0715, 2019-10-11:
> Public comment from Ho.Yan in [Zendesk ticket #28662](https://djisdksupport.zendesk.com/agent/tickets/28662):
>
> Dear Developer,
>
> Thanks for reaching Ryze!
>
> Please check if there is a [...]

Hi, after installing the getModels.sh file, it finally works. However, I have turned on the Pose Recognition on the UI interface, but it does not work. I have also checked that I have pose_iter_160000.caffemodel inside the mpi path.

May I know whats wrong ?

hkpc0715, 2019-10-11:
> Public comment from Ho.Yan in [Zendesk ticket #28662](https://djisdksupport.zendesk.com/agent/tickets/28662):
>
> Dear Developer,
>
> Thanks for reaching Ryze!
>
> Please check if there is a [...]

This is the error that I have got in terminal:

[ INFO:0] Initialize OpenCL runtime...
OpenCV Error: Assertion failed (output_slice.isContinuous() && output_slice.size == curr_output.size) in fuseLayers, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/dnn/src/dnn.cpp, line 1428
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/spark/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello_control_ui.py", line 112, in videoLoop
    cmd,self.draw_skeleton_flag,self.points = self.my_tello_pose.detect(self.frame)
  File "/home/spark/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello_pose.py", line 209, in detect
    output = self.net.forward()
error: /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/dnn/src/dnn.cpp:1428: error: (-215) output_slice.isContinuous() && output_slice.size == curr_output.size in function fuseLayers

dji-dev, 2019-10-11:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/28662'>Zendesk ticket #28662</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

You can test first ，if you have any questions, please contact us.

Best Regards,
Ho Yan
RYZE TECHNICAL SUPPORT

hkpc0715, 2019-10-11:
> Public comment from Ho.Yan in [Zendesk ticket #28662](https://djisdksupport.zendesk.com/agent/tickets/28662):
> 
> Dear Developer,
> 
> Thanks for reaching Ryze!
> 
> You can test first ，if you have any questions, please contact us.
> 
> Best Regards,
> Ho Yan
> RYZE TECHNICAL SUPPORT

Hi Ho.Yan,

I have just tested and got the following error:

This is the error that I have got in terminal:

```
[ INFO:0] Initialize OpenCL runtime...
OpenCV Error: Assertion failed (output_slice.isContinuous() && output_slice.size == curr_output.size) in fuseLayers, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/dnn/src/dnn.cpp, line 1428
Exception in thread Thread-3:
Traceback (most recent call last):
File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
self.run()
File "/usr/lib/python2.7/threading.py", line 754, in run
self.__target(*self.__args, **self.__kwargs)
File "/home/spark/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello_control_ui.py", line 112, in videoLoop
cmd,self.draw_skeleton_flag,self.points = self.my_tello_pose.detect(self.frame)
File "/home/spark/Downloads/Tello-Python-master/Tello_Video_With_Pose_Recognition/tello_pose.py", line 209, in detect
output = self.net.forward()
error: /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/dnn/src/dnn.cpp:1428: error: (-215) output_slice.isContinuous() && output_slice.size == curr_output.size in function fuseLayers
```

huaerduoduokaibuting, 2020-10-08:
@hkpc0715   do  you solve this problem?   help!

frank804, 2020-10-08:
@hkpc0715 do you solve this problem? help!