dji-sdk/Tello-Python #42: Tello_Video & Tello_Video_With_Pose_Recognition #41

Momo-AS, 2019-10-27:
Hi all,

I was trying out both Tello_Video & Tello_Video_With_Pose_Recognition packages in my Mac Os catalina
At first, I installed Tello_Video with the one click install and it not work. still getting the error: 

CMake Error at /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package):
  Could not find a package configuration file provided by "boost_python"
  (requested version 1.71.0) with any of the following names:

    boost_pythonConfig.cmake
    boost_python-config.cmake

  Add the installation prefix of "boost_python" to CMAKE_PREFIX_PATH or set
  "boost_python_DIR" to a directory containing one of the above files.  If
  "boost_python" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/local/lib/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/local/Cellar/cmake/3.15.4/share/cmake/Modules/FindBoost.cmake:443 (find_package)
  CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Tello_Video/h264decoder/build/CMakeFiles/CMakeOutput.log".

Any help will be greatly appreciated.

lmmedino, 2019-11-19:
I am having the same error, have you had any progress?

tomlewandowski, 2019-11-20:
I have the same issue, anyone has solution?

lmmedino, 2019-11-21:
I was able to compile by replacing the cmake line by this "cmake -D Boost_NO_BOOST_CMAKE:BOOL=ON .."

ddanielorlano, 2019-12-26:
Add this line `SET(Boost_NO_BOOST_CMAKE ON CACHE BOOL "Enable fix for FindBoost.cmake")`
To line 114 of `/usr/local/Cellar/boost/1.71.0/lib/cmake/Boost-1.71.0/BoostConfig.cmake`

So it now looks like this:
```
  SET(Boost_NO_BOOST_CMAKE ON CACHE BOOL "Enable fix for FindBoost.cmake")
  if(Boost_DEBUG)
    message(STATUS "BoostConfig: find_package(boost_${__boost_comp_nv} ${Boost_VERSION} EXACT CONFIG ${_BOOST_REQUIRED} ${_BOOST_QUIET} HINTS ${_BOOST_CMAKEDIR})")
  endif()
  find_package(boost_${__boost_comp_nv} ${Boost_VERSION} EXACT CONFIG ${_BOOST_REQUIRED} ${_BOOST_QUIET} HINTS ${_BOOST_CMAKEDIR})
```