dji-sdk/Tello-Python #43: I am having a similar issue as others with the h264decoder

Dwhyte62, 2019-10-28:
I am trying to use the tello drones in a school environment and can not get video working. We are using python 3.7 in the anaconda. Could this be part of the problem? Is there an update for later versions on python?

dji-dev, 2019-10-29:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/29278'>Zendesk ticket #29278</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

Please confirm that your development environment is fully configured：
https://github.com/dji-sdk/Tello-Python/blob/master/TelloPython_FAQ.txt

Best Regards,
Ho Yan
RYZE TECHNICAL SUPPORT

Dwhyte62, 2019-10-29:
I have followed all of the steps suggested using the zip and installing the required libraries. The h264decoder is pasted into C:\Program Files\Python37\Lib\site-packages. The only library that was not included in anaconda is opencv-python which I installed. I still get the error with import libh264decoder

dji-dev, 2019-10-30:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/29278'>Zendesk ticket #29278</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

It is recommended that you use the terminal to implement
 [image]​

Best Regards,
Ho Yan
RYZE TECHNICAL SUPPORT<br><a href='https://djisdksupport.zendesk.com/attachments/token/Ght2RwcGSMj5c7XFQthsfOHjm' target='zd_attach'>inline-1649186874.png</a>

xtr33me, 2019-10-30:
I too am dealing with this same error in Ubuntu 18.04, using a virtualenv and python 3.6. I compiled from source and copied the .so file into my virtualenv site-packages directory. I still get "No module named libh264decoder".

theman162, 2019-10-31:
hi,i do not have the  file libh264decoder.so during installation in widows

dji-dev, 2019-10-31:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/29278'>Zendesk ticket #29278</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

Attached is a libh264decoder file, please download to get.​

Best Regards,
Ho Yan
RYZE TECHNICAL SUPPORT<br><a href='https://djisdksupport.zendesk.com/attachments/token/i85lC3HkUIKYFecF22wHT3d1j' target='zd_attach'>h264decoder.7z</a>

xtr33me, 2019-10-31:
Thank you Ho, but it seems I am still dealing with the issue.

Just to be sure, I am copying the file into the following path in my virtualenv:
**~/.virtualenvs/tello/lib/python3.6/site-packages**

[https://ibb.co/qDT9xhh](Screenshot of error in PyCharm)

If you have any other ideas I would love to hear them.  Otherwise, I will let you know if I figure anything out as well. Thanks for your time!

dji-dev, 2019-11-01:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/29278'>Zendesk ticket #29278</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

Environment testing can be used under Python 2.7, as we cannot guarantee that the 3.7 environment is also compatible.

Best Regards,
Ho Yan
RYZE TECHNICAL SUPPORT

biyisi, 2019-11-29:
What kind of operation should be performed on libh264decoder under mac environment?
Show libh264decoder model does not exist in pycharm environment
I tried to copy the libh264decoder folder and libh264decoder.so to site-packages, but I still get an error?

cozmobotics, 2020-01-06:
> Public comment from Ho.Yan in [Zendesk ticket #29278](https://djisdksupport.zendesk.com/agent/tickets/29278):
> 
> Dear Developer,
> 
> Thanks for reaching Ryze!
> 
> Environment testing can be used under Python 2.7, as we cannot guarantee that the 3.7 environment is also compatible.
> 
> Best Regards,
> Ho Yan
> RYZE TECHNICAL SUPPORT

This is a point to respect. 
To all Python 3 users: When the developer says Python 2, he means Python 2. 
Python 3 is not downward-compatible to Python 2. 
Does it work with Python 2 ? 

However, I have tried it with Python 2.7 and was not successful, either.

francishandc, 2020-09-26:
I'm looking for h264decoder.pyd (instead of lib264decoder.pyd) to run on Python3. May I know where I can download it?
Thanks!