dji-sdk/Tello-Python #48: no responce in the terminal after running main.py in tello video

ShravanAnandk7, 2019-11-25:
I have installed all the libraries allowed UDP 11111 port on the firewall, connected to wifi successfully and run the code in terminal on python2.7 but there is no responce....

dji-dev, 2019-11-26:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/30153'>Zendesk ticket #30153</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

Is there any information that indicates this errors?
Also, make sure that the aircraft is fully charged and that you have successfully installed all the environments.

Thanks and have a nice day!

Regards,
RYZE TECHNICAL SUPPORT

MaikGon, 2021-05-02:
Hi, the drone does not respond to my commands too. I run the code (main.py) in the terminal, I can see the video stream, but the drone doesn't execute my commands. There is no error information from the drone.