dji-sdk/Tello-Python #49: What kind of operation should be performed on libh264decoder under mac environment?

biyisi, 2019-11-29:
What kind of operation should be performed on libh264decoder under mac environment?
Show libh264decoder model does not exist in pycharm environment
I tried to copy the libh264decoder folder and libh264decoder.so to site-packages, but I still get an error?

biyisi, 2019-11-29:
In pycharm environment, tello.py code tells no module named "libh264decoder"

dji-dev, 2019-11-29:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/30322'>Zendesk ticket #30322</a>:<br><br>Dear developer,

Hello, thank you for your support.

If you have to compile it from source,with Linux or Mac,you can:

    cd h264decodermkdir buildcd buildcmake ..makecp libh264decoder.so ../../

Regards,

Ho Yan
RYZE TECHNICAL SUPPORT