dji-sdk/Tello-Python #50: errors about video transmission

Githubwujinming, 2019-12-03:
runned in raspberrypi , python2.7 , environments descriped in ./linux_install.sh were installed.
Getting the following error. 
[image]

1.error while decoding MB 22 11, bytestream -12
2.non-existing pps 0 referenced

dji-dev, 2019-12-04:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/30418'>Zendesk ticket #30418</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

Have you modified the original code?

Thanks and have a nice day!

Regards,
RYZE TECHNICAL SUPPORT

Githubwujinming, 2019-12-05:
Thank you for your response, it works on PC without error,  raspberrypi may not power enough to real-time decode h264 package , so that only  part of the image was decoded.  is there any possibility to make it work on raspberrypi ? any suggestions are welcome.

dji-dev, 2019-12-09:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/30418'>Zendesk ticket #30418</a>:<br><br>Dear Developer,

Thanks for reaching Ryze!

Raspberrypi is not our official recommendation and may require you to test it yourself.

Thanks and have a nice day!

Regards,
RYZE TECHNICAL SUPPORT

closed 2023-03-02 by Githubwujinming