dji-sdk/Tello-Python #55: there is no video from tello edu...

tt503177177, 2019-12-17:
hi all could someone please help me? i use ubuntu 16.04

i copy Tello_Video

install   linux_install.sh

use the step from demend to compile h264decoder

and python main.py 

but there is no video GUI show up?

Please tell me how to solve it.....is there something step I forgot?

dji-dev, 2019-12-19:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/30876'>Zendesk ticket #30876</a>:<br><br>Dear developer,

Hello, thank you for your feedback.

Under normal circumstances, in the case of tello wifi is already connected, the GUI will pop up the video stream after running the program, and can use the snapshot button normally to take pictures if it is not working properly
  Consider and check for the following:
  a. Whether your PC device is properly connected to your tello wifi;
  b. Check your PC's network firewall, which may block video streaming data;
  c. Further check your network settings, because tello is the PC's 11111 port video stream data push, so check to see if your network settings disable the 11111UDP port, or the 11111 port of UDP data was blocked.
    If so, undisable;
  d. In rare cases, it is possible that the 'streamon' instruction sent by the PC has lost the packet and has not been received by tello, for which case, try re-running the program.

Thank you very much for your understanding and cooperation, I wish you a happy life!

Regards,
RYZE TECHNICAL SUPPORT

tt503177177, 2019-12-19:
thanks but i had seen this doucument from ur readme
so i want  know is there another way to solve?
please help meQQQQQQQQQQQQQ

dji-dev, 2019-12-20:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/30876'>Zendesk ticket #30876</a>:<br><br>Dear developer,

Hello, thank you for your feedback.

Please tell me to have any error tips?

Thank you very much for your understanding and cooperation, I wish you a happy life!
Regards,
RYZE TECHNICAL SUPPORT

ZGX010, 2020-01-06:
I also face this problem , and i setting the network firewall off, i get video streaming data.

luccosta, 2020-01-29:
I had the same problem and after enable all the  UDP ports in the network firewall the problem stills the same.

All the dependencies are properly installed and like in the other cases, the program run, open the Tkinter GUI, has no stream and the snapshot cause an error in the OpenCV program that I think is the lack of image in the stream.

I'm on Windows 10 in Portuguese. Follows the image of the permission for all UDP ports.

[image]

luccosta, 2020-02-11:
I resolve the problem updating the Tello firmware in the app.

suet-lee, 2020-02-12:
@luccosta Hi, what version of the firmware did you update to? I have the 02.04.90.01 version and the video stream is still not working

tt503177177, 2020-02-23:
oh i solve this by also update in the app  and i find that it is not work in the virsualBox (to use linux)
after I install ubuntu it solved

tt503177177, 2020-02-23:
I aso

tt503177177, 2020-02-23:
I also solve bu update in app  and  i find to it is not work in the virtualbox with linux
I install ubuntu in my pc and it solved

tkin91e55, 2020-08-17:
The Tello_Video main.py had died at start, the problem was on video streaming. I disabled it and the UI buttons worked only partly, there had been no video streaming popped out in Tello_Video/Tello_Video_Pose examples. I solved it.

My case was on Mac, macOS Catalina. There were a number of settings in mac_install.sh outdated to run at Python2.7.
I compiled the libh264decoder.so, further investigation took me to the line "self.decoder.decode(packet_data)", which did not worked well with ffmpeg (version 4), I brew install ffmpeg@2.8 instead.
Points I checked:

(1) ffmpeg need to be older, I got 2.8 worked. Only installed ffmpeg@2.8 doesn't work out of the box, need to tell the python, the .so file to direct to ffmpeg@2.8
(2) the opencv-python used was 3.4.5.20
(3) the pose-recognition link was dead

Refer to https://github.com/tkin91e55/Tello-Python/blob/master/Tello_Video/install/Mac/mac_install.sh