dji-sdk/Tello-Python #57: Python 3 (again)

f41ardu, 2019-12-28:
Any thoughts to migrate the Tello-Python code to Python 3.X 

I managed to migrate the H264 that it can be used with Python 3.X on my Raspberry PI. 
The rest is straightforward. So if someone could help to build the H264 dll or pyc from my fork could be helpful.

RahilRehan, 2019-12-28:
I'm a beginner, but I would like to help. I want this project to be in python 3 for my project.

f41ardu, 2019-12-29:
Stay tuned. I've nearly finalised it either on Raspberry x86 implementation. The implementation on my native Raspberry isn't completed yet.  Be aware, video performance on a Raspberry PI3+ isn't perfect. But I assume that it will work on Linux, too. See my fork for further details.

dji-dev, 2019-12-30:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/31175'>Zendesk ticket #31175</a>:<br><br>Dear Developer,
Thank you for contacting DJI.

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

woofwoof, 2020-01-08:
Hi,

Thanks for trying to port this to Python 3.X. I tried to install the original 2.7 version but received a warning that 2.7 is no longer supported (as of Jan 1st 2020, apparently). I'd like to get video_test up and running. I can help test for you but I'm probably not as good a Python programmer as yourself! :)

I saw @f41ardu that you are experiencing the decode problems - does this mean that the video_test isn't working yet in Python 3.X please? To be honest I don't think I really need video but would more likely need a still image from the stream...

Thanks again. WW.

woofwoof, 2020-01-08:
I should have also mentioned that the 'one click install' couldn't install openCV either - not sure if that's related to Python 2.7 or something else. I haven't tried a manual install just yet though. Cheers, WW.

f41ardu, 2020-01-18:
My fork works fine using Python3, only on PI 3+ is somewhat slow. 
But there is a wrong path in my README for the boost lib for Raspberry. Stay tuned I'll update soon. 
Do we have information on the performance in Ubuntu or ???

f41ardu, 2020-01-22:
udp video stream can be encoded by opencv:

https://gist.github.com/f41ardu/b75da46dae383e5c835295600deef604

No need to use the h264 lib any longer. Migration to Phyton3 could become fairly easy.

jstolp, 2020-02-25:
So can we run this with Python3 or not? I'm sure I can manage. But any official support would be appreciated.

justingrayston, 2022-06-23:
Any update on a Python3 version or progress of one we could create a PR. It looks a bit like this repo is abandoned by DJI.

pitfloyd, 2022-08-25:
very latancy with video stream how remove the latency?

nobu835, 2024-10-08:
(The last update was two years ago, so maybe no one is interested in this,)
The easiest way to play Tello videos with the lowest latency using Python 3 is to use the h264decoder:
https://github.com/DaWelter/h264decoder

If you are using windows you will need to replace some of the code. Here is a guide I wrote.
https://github.com/meekworth/pylwdrone/issues/10#issuecomment-2396154826

Replace `import libh264decoder` in tello.py with `import h264decoder`.