dji-sdk/Tello-Python #59: Python 2.7 EOL

woofwoof, 2020-01-08:
Hi,

Python 2.7 has reached its End Of Life. Its at least no longer supported. OpenCV won't install using pip (on a raspberry pi at least) even from the additional repository. Pip3 installs OpenCV (quite rightly) in a Python3 directory on a Raspberry Pi but that's incompatible for this code: "import cv2" fails as its in the wrong directory? I'm not sure if one could build OpenCV to make it compatible with Python2.7?

So, as far as I can see the best solution all round would be to port this code to Python 3 as others are saying please? I don't know how big a job that is though?

Cheers, WW.

dji-dev, 2020-01-10:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/31432'>Zendesk ticket #31432</a>:<br><br>Dear Developer,
Thank you for contacting DJI.

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

baqwas, 2021-01-10:
Apologies for latching on to a year old thread. I've just purchased a Tello Drone and haven't quite started on programming it. While looking for answers, I stumbled upon this repository and just want to assist any way I can.

Instructions on building OpenCV are available at [https://docs.opencv.org/master/df/d65/tutorial_table_of_content_introduction.html](https://docs.opencv.org/master/df/d65/tutorial_table_of_content_introduction.html)
Drop me a note if you need assistance in building on Ubuntu or Raspberry Pi for Python3 or C++.

Kind regards.

woofwoof, 2021-01-18:
Hi. Thanks. I believe I concluded the TelloPy code needed to be in Python 3 really. Python 2.7 is at end of life. Could you help port it please? Or is that too big a job? To be honest I haven't looked at this recently but I hope to get back to it. Best wishes. WW. P.S. I just checked my old notes & I got as far as needing to install AV but that had further dependencies of ffmpeg etc. I guess I gave up & moved on at that point! 😊

baqwas, 2021-01-19:
Hello @woofwoof,
I'll be using a Tello EDU (black) drone for the exercise. I will start around mid-Fed on the project. Trying to get all my ducks in a row (as the saying goes).

As for basic instructions on building OpenCV with the latest GitHub master, here are the basic instructions including some redundant prerequisites:

`sudo apt-get install build-essential
sudo apt-get install cmake git libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev
sudo apt-get install python-dev python-numpy libtbb2 libtbb-dev libjpeg-dev libpng-dev libtiff-dev libjasper-dev libdc1394-22-dev

cd ~/projects # optional

git clone https://github.com/opencv/opencv.git

git clone https://github.com/opencv/opencv_contrib.git

git clone https://github.com/opencv/opencv_extra.git

sudo apt -y install build-essential cmake pkg-config libpng12-0 libpng12-dev libpng-dev libpng++-dev libpnglite-dev zlib1g-dbg zlib1g zlib1g-dev pngtools libtiff5-dev libtiff5 libtiffxx0c2 libtiff-tools

sudo apt -y install libjpeg8 libjpeg8-dev libjpeg8-dbg libjpeg-progs ffmpeg libavcodec-dev libavcodec57 libavformat57 libavformat-dev libgstreamer0.10-0-dbg libgstreamer0.10-0 libgstreamer0.10-dev libxine2-dev libunicap2 libunicap2-dev libdc1394-22-dev libdc1394-22 libdc1394-utils swig libv4l-0 libv4l-dev python3-numpy python3-dev libgtk2.0-dev pkg-config

sudo apt install -y cmake-qt-gui libatlas-base-dev gfortran pylintss libblas-dev liblapack-dev libxine1-ffmpeg libxine1-bin 

cmake -D CMAKE_BUILD_TYPE=Release -D CMAKE_INSTALL_PREFIX=/usr/local -D PYTHON_DEFAULT_EXECUTABLE=/usr/bin/python3.5 -D ENABLE_PRECOMPILED_HEADERS=OFF ..

For Raspberry Pi older models:
Change the size in /etc/dphys-swapfile
CONF_SWAPSIZE=1024

If you want to change the size, you need to modify the number and restart dphys-swapfile:
sudo dphys-swapfile setup
sudo /etc/init.d/dphys-swapfile stop
sudo /etc/init.d/dphys-swapfile start
/etc/init.d/dphys-swapfile restart

sudo apt -y install build-essential cmake pkg-config libpng-dev libpng++-dev libpnglite-dev zlib1g zlib1g-dev pngtools libtiff5-dev libtiff5 libtiff-tools

sudo apt -y install libjpeg8 libjpeg8-dev libjpeg8-dbg libjpeg-progs ffmpeg libavcodec-dev libavformat-dev libxine2-dev libunicap2 libunicap2-dev swig libv4l-0 libv4l-dev python3-numpy python3-dev libgtk2.0-dev pkg-config

sudo apt install -y cmake-qt-gui libatlas-base-dev gfortran pylint libblas-dev liblapack-dev
libxine1-ffmpeg libxine1-bin

sudo apt install -y libcanberra-gtk*

cd opencv

mkdir build

cd build

ccmake ..

cmake .

nproc –all

make -j$(nproc)

sudo make install`

woofwoof, 2021-01-19:
Hello again. Like I said, I previously concluded the issue is that tellopy now needs to be in Python 3 see e.g. https://www.python.org/doc/sunset-python-2/. I appreciate your input but I think you're missing my point. Best, WW.