dji-sdk/Tello-Python #60: how to reduce the delay of video stream

alexmihalyk23, 2020-01-12:
I'am use opencv and imutils library's for capture video(imutils - filevideostream), but get delay on stream. How to fix this?

dji-dev, 2020-01-14:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/31542'>Zendesk ticket #31542</a>:<br><br>Dear Developer,
Thank you for contacting DJI.

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

alexmihalyk23, 2020-01-24:
I'am fixed it using WebcamVideostream

closed 2020-01-24 by alexmihalyk23

manujimele, 2020-05-27:
> I'am fixed it using WebcamVideostream

How have you done that?

I have issues trying to get frames from the camera.
Any info? any help?

luohc2004, 2020-09-22:
Can I obtain the picture taken from the down side bottom view camera?

alexmihalyk23, 2020-09-30:
@Manujimele https://github.com/jrosebr1/imutils/blob/master/imutils/video/webcamvideostream.py

Marselias, 2021-02-08:
Does anyone have a better solution? WebcamVideostream only provides threading, which wasn't helpful for me.