dji-sdk/Tello-Python #62: Could not find a package configuration file provided by "boost_python"

Wendy-hot-cpu, 2020-01-21:
I tried the One-Click installization
but something wrong with cmake.
by the way I use Mac
[image]

dji-dev, 2020-01-22:
Public comment from Lisa Fedane in <a href='https://djisdksupport.zendesk.com/agent/tickets/31771'>Zendesk ticket #31771</a>:<br><br>DJI Developer Support engineers will be on holiday for the Chinese Spring Festival (1-23 ~ 1-30), which may cause some delays in responding to your issue. We appreciate your patience and apologize for the inconvenience this may cause you.

Thank you for you understanding and we will answer this as quickly as possible.

dji-dev, 2020-01-27:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/31771'>Zendesk ticket #31771</a>:<br><br>Dear Developer,
Thank you for contacting DJI.

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

ryanxjhan, 2020-05-06:
Same issue here