dji-sdk/Tello-Python #64: with other hotspot , no video?(in python)

tt503177177, 2020-02-23:
Hi all ! could plz help me solve the problem?
I use  

ap  'wifi name'  'passward'   ( i also chek the tello really ip state in my code)

command
streamon
ffplay -f h264 udp://0.0.0.0:11111

and not video show? 
and i find this  
"Q8:Can Tello EDU transmit video streams externally in station mode (send ap 'ssid' 'password' command for Tello EDU)?
A8:This feature is currently not supported."             at

https://github.com/dji-sdk/Tello-Python/blob/693776d65b691525773adf4f320f034946fb15b2/TelloPython_FAQ.txt

is it mean there no way to solve...? 
can someone success by other way ? could tell me please, thx

dji-dev, 2020-02-24:
Public comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/32312'>Zendesk ticket #32312</a>:<br><br>Dear Developer,
Thank you for contacting DJI.

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

ghost-ronin, 2020-03-02:
> Hi all ! could plz help me solve the problem? (My drone is Tello EDU)
> I use
> 
> ap 'wifi name' 'passward' ( i also chek the tello really ip state in my code)
> 
> command
> streamon
> ffplay -f h264 udp://0.0.0.0:11111
> 
> and not video show?
> and i find this
> "Q8:Can Tello EDU transmit video streams externally in station mode (send ap 'ssid' 'password' command for Tello EDU)?
> A8:This feature is currently not supported." at
> 
> https://github.com/dji-sdk/Tello-Python/blob/693776d65b691525773adf4f320f034946fb15b2/TelloPython_FAQ.txt
> 
> is it mean there no way to solve...?
> can someone success by other way ? could tell me please, thx

hi brother.Can you try create UDP traffic point 'for video 11111' on your computer?you should this page for add new rule.
https://docs.microsoft.com/en-us/windows/security/threat-protection/windows-firewall/create-an-inbound-port-rule