dji-sdk/Tello-Python #65: what is 'dm'?   python tello send 'height?' problem

tt503177177, 2020-04-14:
hi all! could someone please help with this problem？

if I type 'height?'
tello will  return  0dm(before takeoff)

if takeoff  i sent 'height?'
i will return  1dm  2dm 3dm.........
what is 'dm'?
is it mean 'Relative height'?
and how let it print 'now at 1m' 'now at 2m'.....?

dji-dev, 2020-04-14:
Agent comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/33533'>Zendesk ticket #33533</a>:<br><br>Dear Developer,
Thank you for contacting DJI.

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

guowanhui, 2020-04-16:
Tello takes take-off point as 0 altitude发自我的华为平板电脑-------- 原始邮件 --------发件人： tt503177177 <notifications@github.com>日期： 2020年4月14日周二 下午2:54收件人： dji-sdk/Tello-Python <Tello-Python@noreply.github.com>抄送： Subscribed <subscribed@noreply.github.com>主    题： [dji-sdk/Tello-Python] what is 'dm'?   python tello send 'height?' problem (#65)
hi all! could someone please help with this problem？
if I type 'height?'
tello will  return  0dm(before takeoff)
if takeoff  i sent 'height?'
i will return  1dm  2dm 3dm.........
what is 'dm'?
is it mean 'Relative height'?
and how let it print 'now at 1m' 'now at 2m'.....?

RobertoMinelli, 2021-03-24:
Dear @tt503177177,

Tello returns **relative height** from takeoff point (which corresponds to 0dm).
`dm` stands for **decimeter** — a unit of length in the metric system, equal to one tenth of a metre.

Cheers,
Roberto