dji-sdk/Tello-Python #66: 【tello pose reconition】Can't open "model/pose/mpi/pose_deploy_linevec_faster_4_stages.prototxt" in function 'cv::dnn::ReadProtoFromTextFile'

amixlong, 2020-04-29:
cv2.error: OpenCV(3.4.2) C:\projects\opencv-python\opencv\modules\dnn\src\caffe\caffe_io.cpp:1121: error: (-2:Unspecified error) FAILED: fs.is_open(). Can't open "model/pose/mpi/pose_deploy_linevec_faster_4_stages.prototxt" in function 'cv::dnn::ReadProtoFromTextFile'

amixlong, 2020-04-29:
I solved it,by hanging the relative path to absolute path.

amixlong, 2020-04-29:
I solved it,by changing the relative path to absolute path.

dji-dev, 2020-04-29:
Agent comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/34062'>Zendesk ticket #34062</a>:<br><br>Dear Developer,
Thank you for contacting DJI.

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

TejasTidke, 2020-05-12:
> 
> 
> I solved it,by changing the relative path to absolute path.

@amixlong 
I'm also facing the same error, can you explain how did you resolved it?

amixlong, 2020-05-15:
In the file of  tello_pose.py,I changed weightsFile and protoFile 's paths as absolute paths ,then it worked.Hope for this reply can help you.
| |
龙树燏
|
|
longshuyu21@126.com
|
签名由网易邮箱大师定制