dji-sdk/Tello-Python #68: ffmpeg isn't available anymore

EmSavalle, 2020-05-25:
Hi
I've tried using the SDK on Windows, and installing everything with the .bat file.
But it says the ffmpeg isn't available, and searching on the archives, I've seen that the version use by this sdk isn't available anymore?
Is there a fix for it?

Thank you,
Emile

dji-dev, 2020-05-29:
Dear Developer,

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

jamespschubert, 2020-08-19:
So was this ever resolved? It sort of makes this entire repository useless with out it and to just say contact Ryze tello support is sort of stupid and useless. WHERE IS THE FFMPEG file? Should someone fix this issue? The install.bat is useless as is this entire repository./?????????????????????????????????????

JackyWu311, 2020-12-01:
Download this project and unzipped "tello_video_dll(ForWin64).zip".
I think all files is in the unzipped folder.
[image]