dji-sdk/Tello-Python #69: Tello_video not working in Ubuntu 20.04

pczekalski, 2020-05-31:
I'm trying to run TelloVideo under Ubuntu 20.04 LTS with no success.

While importing libh264decoder (compilation went OK, just some warnings), I got:

```
Exception has occurred: ImportError
/lib/x86_64-linux-gnu/libboost_python38.so.1.71.0: undefined symbol: _Py_tracemalloc_config
  File "/home/pczekalski/Documents/Repos/Tello-Python/Tello_Video/tello.py", line 5, in <module>
    import libh264decoder
```
The python version is 2.7.18.rc1

Also note, Ubuntu 20.04 no longer supports pip on python2 - you need to refer to it as pip2 in the installation script.

Regards,

Piotr

dji-dev, 2020-06-01:
Agent comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/35260'>Zendesk ticket #35260</a>:<br><br>Dear Developer,

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

snakehaihai, 2020-09-21:
> I'm trying to run TelloVideo under Ubuntu 20.04 LTS with no success.
>
> While importing libh264decoder (compilation went OK, just some warnings), I got:
>
> ```
> Exception has occurred: ImportError [...]

manually link the import.  by calling import sys above. then sys.path.insert(0, PATHTOYOURH264)
then import libh264decoder. this worked for me

pi0projects, 2021-07-26:
hey @snakehaihai, could you kindly share your script on how you make it work, I am new to programming. I also am running ubuntu 20.04

thank you! 👊

snakehaihai, 2021-07-28:
> hey @snakehaihai, could you kindly share your script on how you made it work. I am new to programming. I also am running ubuntu 20.04
> 
> thank you!

I end up giving up this code. 

To get the video up, just use udp packet sender tools to send. 

to real-time subscribe the video. just use the ffplay tools to see. or using ffmepg to capture to OpenCV

You may follow up my post here to get things up https://stackoverflow.com/questions/67796536/iptables-routing-packets-with-a-raspberry-pi-to-a-tello-drone/67809682#67809682

pi0projects, 2021-07-29:
thanks! i will check it out

dicisoft, 2021-12-30:
> thanks! i will check it out

@pi0projects finally you solved this issue ?

allrobot, 2022-03-25:
@pi0projects @pczekalski 
Hello, how are you solving it?

pczekalski, 2022-03-25:
Hi,

Not at all. Just switched to other drones with PX4 stack.

Regards,

P.