dji-sdk/Tello-Python #70: ImportError: No module named libh264decoder

mikeyhao, 2020-07-18:
I've seen this issue around for awhile now but still no fix to it. I read the FAQ and tried manually installing and also installing through the vcredist_x64.exe file too. I have attached how my site packages file looks like in my python folder for reference. 
[image] , you can ignore the .cpp/ .hpp files I added them in to see if anything would change. Is there a workaround or solution found?

dji-dev, 2020-07-20:
Agent comment from Ho.Yan in <a href='https://djisdksupport.zendesk.com/agent/tickets/37067'>Zendesk ticket #37067</a>:<br><br>Dear Developer,
Thank you for contacting DJI.

This question you can contact RYZE Tello's technical support：
support.cn@ryzerobotics.com

Warm Regards,
DJI Developer Support

jere-fade, 2021-01-31:
It seems that I have encountered the same problem
[image]

dg022, 2021-05-30:
Has anyone found a solution to this, Im getting the same error

EdwardUni22, 2024-03-01:
did anyone amange to fix this yet?