dji-sdk/Tello-Python #85: Tello EDU not returning ok/error for "emergency" command

anoop4, 2022-08-03:
The Tello SDK documentation mentions that the Tello EDU returns an "ok/error" response for the "emergency" command.
https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20SDK%202.0%20User%20Guide.pdf

However, I see that my EDU drone doesn't return an "ok" response even though it emergency lands appropriately.

Is this a known issue?

dji-dev, 2022-08-03:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/68652'>Zendesk ticket #68652</a>:<br><br>Hi this is known issue. In fact some commands do not return. Please refer to the actual test results.
<br><p><em><small>°°°</small></em></p>

anoop4, 2022-08-03:
Hi, I am not able to view the ticket you linked in the previous reply.

Can you let me know which Firmware version did this issue originate?
Also, is there an expected ETA for the fix?

dji-dev, 2022-08-03:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/68652'>Zendesk ticket #68652</a>:<br><br>This issue always exist. This problem occurs because UDP connections have a high probability of packet loss. And some commands have relatively serious packet loss. I am sorry we have no plan to optimize this problem. Is this problem blocking your development process?
<br><p><em><small>°°°</small></em></p>

anoop4, 2022-08-03:
Yes, this is a blocking issue for our application.

We rely on the command's return value to determine if a particular command execution was successful. In this case, the emergency command (and possibly other commands) doesn't return "ok"/"error" and there is no way to determine if the emergency land was successful.
As a result, we are not able to give appropriate feedback to the user from the application.

The situation is even critical for the basic movement commands (up, down, left, right, forward, back) since there is no way to programmatically determine if the command was successful. The only way t validate this is by visual inspection.

anoop4, 2022-08-09:
Is there any update on this issue?