dji-sdk/Tello-Python #87: video latency

mtncgj123, 2022-08-10:
Hello, this project seems good, but when I run the code, there is serious video latency, the image will blur for 2 seconds and restore and blur again with the error below
[image]
 can anyone helps?

pitfloyd, 2022-08-25:
hii 
you resolve?

mtncgj123, 2022-08-26:
@juanma-t3j  Yeah, I tried to use the latest robotmaster SDK (https://github.com/dji-sdk/RoboMaster-SDK) and received more stable video stream

1411734304, 2024-04-07:
@mtncgj123 Can I have a look at your code? I want to learn it, because my current code ability is very weak。
Thank you very much and look forward to your reply。

mtncgj123, 2024-04-07:
> @mtncgj123 Can I have a look at your code? I want to learn it, because my current code ability is very weak。 Thank you very much and look forward to your reply。

Sure, you can leave me with your email address or contact me with my email in my profile

1411734304, 2024-04-07:
@mtncgj123 My email address is 1411734304@qq.com. Thank you very much for your help. If it doesn't work, you can try jingqihu38@gmail.com