dji-sdk/Tello-Python #88: Run timeout

ilyak93, 2022-08-28:
Hi, I get this error `response: error Run timeout` each time I try run each of the commands:

`go 80 0 0 20 m1` or `jump 80 0 0 20 90 m1 m2`

this is my commands file:

```
command
mon
delay 1
takeoff
delay 5
go 80 0 0 20 m1
delay 5
land
```

and this is how the log looks like:
```

id: 0
command: command
response: ok
start time: 2022-08-28 19:48:22.754000
end_time: 2022-08-28 19:48:22.782000
duration: 0.028

id: 1
command: mon
response: ok
start time: 2022-08-28 19:48:22.782000
end_time: 2022-08-28 19:48:22.829000
duration: 0.047

id: 2
command: takeoff
response: ok
start time: 2022-08-28 19:48:23.836000
end_time: 2022-08-28 19:48:27.725000
duration: 3.889

id: 3
command: go 80 0 0 20 m1
response: error Run timeout
start time: 2022-08-28 19:48:32.738000
end_time: 2022-08-28 19:48:47.921000
duration: 15.183

id: 4
command: land
response: ok
start time: 2022-08-28 19:48:52.745000
end_time: 2022-08-28 19:48:54.407000
duration: 1.662

```
I tried to increase and decrease the delays, didn't correct the behavior.

The delay command as much as I understand should be set according to the duration of the command execution mentioned in the log, am I right ?

I.e the delay should be >= of the duration of the command executed before the delay command am I right ?
e.g:

duration of  `go 80 0 0 20 m1` command is `duration: 15.183` so the commands should look like:

```
mon
delay 1
takeoff
delay 5
go 80 0 0 20 m1
delay 20
land
```
as the last delay command `delay 20` where `20 >= 15.183`, is that correct ?

dji-dev, 2022-08-29:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/69826'>Zendesk ticket #69826</a>:<br><br>Do you use Tello or Tello edu?
<br><p><em><small>°°°</small></em></p>

ilyak93, 2022-08-29:
@dji-devת Tello edu.

dji-dev, 2022-08-30:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/69826'>Zendesk ticket #69826</a>:<br><br>I think it's possible that the drone didn't see the challenge card and the command timed out. Please try whether you can obtain the information of the challenge card through the plaintext SDK. Link: https://dl-cdn.ryzerobotics.com/downloads/Tello/Tello%20SDK%202.0%20User%20Guide.pdf.
<br><p><em><small>°°°</small></em></p>

ilyak93, 2022-08-30:
@dji-dev, but then he wouldn't do the movement (I forgot to say that in both cases the error appears after the movement, so the pad recognition is ok) in the x-axis as I commanded it, because in `go x y z m1` and `jump x y z m1 m2` command the `m1` stands for the starting point, for example if I place the drone on pad number 2 and do the commands it will say:

```
id: 3
command: go 80 0 0 20 m2
response: error No valid marker
start time: 2022-08-30 16:20:49.345000
end_time: 2022-08-30 16:20:51.523000
duration: 2.178
```

I also tried to move it forward and check the recognized pad id and the recognition is fine, at some point from returning pad id 1 it changes to returning pad id 2 as expected. 

I've noticed the in jump and go commands the drone goes down the lowest point it can, so maybe in `jump` it has difficulties because of the low height, but still awkward that after `go` command it moves the seem to be right distance and then gets the `run timeout` error.

dji-dev, 2022-09-02:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/69826'>Zendesk ticket #69826</a>:<br><br>Another possibility for this error is that the command is blocked. Do you get the same error when you set a 20s delay
<br><p><em><small>°°°</small></em></p>

ilyak93, 2022-09-02:
@dji-dev, well the problem was much simpler and was my mistake. When i set the z coordinate to be 0, the drone went to the lowest point it can, so although in the first place he did recognized the pad, he faulted in the end of each one of the commands because it couldn't recognize the pad after doing the movement (and he need to do it even in the go command to correct it self in the air with respect to the coordinates you gave him, and in jump obviously with respect to the second pad target). 

And it is also hard to debug, because in some point in jump command I can see it do recognized the second pad, but when it reaches most closely to it, because of the 0 in z, it doesn't recognize the pad and can't finish the command.

closed 2022-09-02 by ilyak93