dji-sdk/Tello-Python #89: tello视频获取

Sanshi6, 2022-10-04:
自己希望在Windows11上重新用python3.7实现视频获取，发现tello发送到端口中的数据长度参差不齐，大概是10-600跳动，相同的代码在Ubuntu18.04/Python3.7环境中可以获取到完整的1460长度的数据包，请问有什么解决方案吗？

Sanshi6, 2022-10-04:
[image]

dji-dev, 2022-10-05:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/71466'>Zendesk ticket #71466</a>:<br><br>请问您是通过什么方式获取视频流的，机型是哪款。
<br><p><em><small>°°°</small></em></p>

Sanshi6, 2022-10-05:
在视频获取方式上没有改动，还是使用Tello-Python库中的方式，UDP协议监听@192.168.10.1:11111进行获取，应该是Windows11系统有问题，相同的代码在Windows10上也成功获取到了1640长度的数据包，我使用的无人机是白色的，具体型号不太会看。

dji-dev, 2022-10-05:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/71466'>Zendesk ticket #71466</a>:<br><br>我们确实没有测试过windows 11，建议您这边还是用windows 10开发。
<br><p><em><small>°°°</small></em></p>

1411734304, 2024-04-07:
@Sanshi6 请问您现在还有相关代码吗，我想学习一下

1411734304, 2024-04-07:
@Sanshi6 请问您现在还有相关代码吗，我想学习一下，感谢啦

Sanshi6, 2024-04-07:
> @Sanshi6 请问您现在还有相关代码吗，我想学习一下，感谢啦

你可以关注一下这个库，https://github.com/Sanshi6/TelloTargetTracking，我在后期把程序放在了Windows10上，它正常运行了。不太清楚现在Windows11是否还存在之前的bug

1411734304, 2024-04-07:
感谢啦