dji-sdk/Tello-Python #91: Can I use existing WiFi network as opposed to Tello's own WiFi?

rogeriobonatti, 2022-12-28:
Hi, I would like to have Tello connect to an existing WiFi network in my building as opposed to connecting to the network that the drone generates. Is that possible to set up? I'm flying over long distances indoors, and the signal strength falls quickly, even if I use one extender.

dji-dev, 2022-12-29:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/75410'>Zendesk ticket #75410</a>:<br><br>Please tell me whether your aircraft model is tello or tello edu. Tello can not connect to WiFi but tello edu can.
<br><p><em><small>°°°</small></em></p>

rogeriobonatti, 2022-12-29:
Thank you for the clarification. My drone is the normal Tello. Since Tello edu can connect to an external wifi then I will buy one.

closed 2022-12-29 by rogeriobonatti

dji-dev, 2022-12-30:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/75410'>Zendesk ticket #75410</a>:<br><br>Yes. Tello edu can connect to an external wifi. But you must do some program to control the aircraft. The official APP does not support control through external wifi.
<br><p><em><small>°°°</small></em></p>

rogeriobonatti, 2023-01-03:
Hi @dji-dev  my Tello EDU arrived. Can you please send me instructions on how to exactly connect the Tello EDU to an external Wifi? So far it looks just like the regular Tello, where I can only connect to the Wifi generated by the drone.

dji-dev, 2023-01-05:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/75410'>Zendesk ticket #75410</a>:<br><br>Please refer to https://robomaster-dev.readthedocs.io/en/latest/python_sdk/connection.html. After tello edu is upgraded to the latest firmware, you can use the robomaster SDK for development.
<br><p><em><small>°°°</small></em></p>