dji-sdk/Tello-Python #97: 采用deltelopy这个库连接tello上传在电脑上的视频呈现蓝色，应该如何解决

1411734304, 2024-04-07:
视频传输的问题如何解决，并且我已经安装了ffmpeg了，另外有人知道如何将yolo与tello结合么

nobu835, 2024-10-07:
Hi

If you're having trouble getting video, try this:
https://github.com/dji-sdk/Tello-Python/issues/98

if you want to use YOLOv8, here is a sample:
(It's a bit simplified and not practical.)
```py
from ultralytics import YOLO
import cv2
cap=cv2.VideoCapture(0) # use webcam 
model=YOLO("yolov8n.pt") # you can also use yolov8s.pt, yolov8s-seg.pt, yolov8s-pose.pt, etc...
while True:
    frame=cap.read()[1]
    results=model(frame)
    for i in range(len(results[0].boxes.xyxy)):
        xyxy=results[0].boxes.xyxy[i].tolist()
        cv2.rectangle(frame,(  int(xyxy[0]),int(xyxy[1])  ),(  int(xyxy[2]),int(xyxy[3])  ),(0,0,255),2)
    cv2.imshow("window",frame)
    cv2.waitKey(1)
```

nobu835, 2024-10-08:
For problems that turn blue, try this:

```py
frame = drone.read()
frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
```